espressif/esp-ali-smartliving #1: 更新SDK ali-smartliving-device-sdk-c v1.6后编译失败

bestkeen, 2020-07-07:
## 1. 开发环境

- 模组或开发板：ESPRESSIF esp-wroom-02d

- esp-ali-smartliving 版本 :  v1.6（https://code.aliyun.com/living_platform/ali-smartliving-device-sdk-c/repository/archive.zip?spm=a2c4g.11186623.2.15.25bd306aMZeLo9&ref=rel_1.6.0）：
或 git clone git@code.aliyun.com:living_platform/ali-smartliving-device-sdk-c.git -b rel_1.6.0

- ESP8266_RTOS_SDK 版本 :  v3.3-rc1（https://github.com/espressif/ESP8266_RTOS_SDK ）：

## 2. 问题描述
编译example-》smart_light 提示错误：
/esp-ali-smartliving/ali-smartliving-device-sdk-c/output/release/lib/libiot_sdk.a(awss_dev_ap.o): In function `wifimgr_process_dev_ap_switchap_request':
(.text.wifimgr_process_dev_ap_switchap_request+0x3fb): undefined reference to `softap_decrypt_password'

### 2.1 复现步骤

1. 编译esp-ali-smartliving sdk 
cd ali-smartliving-device-sdk-c
make reconfig (选择SDK平台 esp8266)
make menuconfig
make 
2. 编译example，smart_light

yuanjianmin, 2020-07-30:
我查看了1.6版本的,这个函数softap_decrypt_password只有头文件定义,没有实现的地方,需要向阿里询问缘由,要不然是编译不过的.该工程下的版本,该函数是有实现的,建议还是使用该版本.

yuanjianmin, 2020-08-05:
@bestkeen 我看了最新的1.6.0的代码,我说的softap_decrypt_password函数阿里的放在libawss_security.a库里面了,所以编译的时候需要链一下.你可以按照我如下修改:
+LIBS += iot_sdk awss_security
+COMPONENT_ADD_LDFLAGS += -L $(COMPONENT_PATH)/ali-smartliving-device-sdk-c/output/release/lib -L $(COMPONENT_PATH)/ali-smartliving-device-sdk-c/lib/xtensa $(addprefix -l,$(LIBS))

修改文件为当前工程最外面的component.mk,如果你用cmake编译,也需要按照链接库的方式修改make文件.

bestkeen, 2020-08-06:
@yuanjianmin 好的，谢谢