espressif/esp-ali-smartliving #4: 编译 ali-smartliving-device-sdk-c 库出错

shun-dgut, 2020-08-05:
----------------------------- 以下请删除 -----------------------------

提交问题时，请先仔细阅读该段提示信息！！

为了更高效的解决 issue 中提交的问题，请按照以下 issue 模板进行提交，请注意以下信息：
1. 提交时请删除这段提示文字；
2. 提交时删除以下由 `//` 开头的文字，并对此做详细补充，如没有特殊信息，可写 `无`。

感谢配合。

----------------------------- 以上请删除 -----------------------------

## 1. 开发环境

- 模组或开发板：
- esp-ali-smartliving 版本（0159374）：
- ESP-IDF 版本（9e70825d1）：

## 2. 问题描述

 编译 ali-smartliving-device-sdk-c 库，在 esp-ali-smartliving 目录下执行：make menuconfig，后提示错误
 错误如下
 root@r0lsec-VirtualBox:/home/r0lsec/aliyun/esp-ali-smartliving/ali-smartliving-device-sdk-c# make menuconfig
./src/protocol/mqtt/Config.in:9:warning: multi-line strings not supported
 make: *** [/home/r0lsec/aliyun/esp-ali-smartliving/ali-smartliving-device-sdk-c/build-rules/_rules-top.mk:204：menuconfig] 错 
 误 1

### 2.1 复现步骤

1. cd  esp-ali-smartliving
2. make menuconfig

yuanjianmin, 2020-08-05:
这个应该是你make的环境有问题.

shun-dgut, 2020-08-05:
> 这个应该是你make的环境有问题.

在虚拟机的linux环境，是指make工具的版本？但是我省略这一步是可以编译代码样例工程的

yuanjianmin, 2020-08-05:
是可以编译代码,但是你这个环境,make menuconfig的图形化配置界面好像有问题,无法通过make menuconfig配置功能,只能通过修改make.settings文件修改配置了.

shun-dgut, 2020-08-07:
> 是可以编译代码,但是你这个环境,make menuconfig的图形化配置界面好像有问题,无法通过make menuconfig配置功能,只能通过修改make.settings文件修改配置了.

这是我的linux 版本和make版本，请问你的make版本是多少
r0lsec@r0lsec-VirtualBox:~/桌面$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 20.04 LTS
Release:	20.04
Codename:	focal
r0lsec@r0lsec-VirtualBox:~/桌面$ make -v
GNU Make 4.2.1
为 x86_64-pc-linux-gnu 编译
Copyright (C) 1988-2016 Free Software Foundation, Inc.
许可证：GPLv3+：GNU 通用公共许可证第 3 版或更新版本<http://gnu.org/licenses/gpl.html>。
本软件是自由软件：您可以自由修改和重新发布它。
在法律允许的范围内没有其他保证。

yuanjianmin, 2020-08-07:
我的比你版本还低,但是不确定为啥你的make menuconfig显示不了.你可以先试试阿里官网的https://code.aliyun.com/living_platform/ali-smartliving-device-sdk-c/ 1.3.0版本,然后试试make menuconfig行不行.

shun-dgut, 2020-08-07:
> 我的比你版本还低,但是不确定为啥你的make menuconfig显示不了.你可以先试试阿里官网的https://code.aliyun.com/living_platform/ali-smartliving-device-sdk-c/ 1.3.0版本,然后试试make menuconfig行不行.

还是错误，是同一个地方错误
make: *** [/home/r0lsec/aliyun/esp-ali-smartliving/ali-smartliving-device-sdk-c/build-rules/_rules-top.mk:202：menuconfig] 错误 1
文件内容如下
menuconfig: prebuilt/ubuntu/bin/kconfig-mconf
    202         $(TOP_Q)$(COMMON_CONFIG_ENV) $^ -s $(TOP_DIR)/Config.in $(if $(TOP_Q),2>/dev/null)

yuanjianmin, 2020-08-07:
你只能找度娘了,或者看一下这个有没有帮助https://blog.csdn.net/qq_34031123/article/details/100901973

HoGC, 2021-03-02:
我也无法编译使用make menuconfig，可以提供默认的make.settings文件吗，以便直接make

yuanjianmin, 2021-03-03:
> 我也无法编译使用make menuconfig，可以提供默认的make.settings文件吗，以便直接make

有默认的配置文件的,可以直接make