espressif/esp-ali-smartliving #6: 提示：wifi: ap_probe_send_over, rest wifi status to disassoc 无法重连

bestkeen, 2020-08-14:
根据本SDK，运行smartlight例程，长时间工作会偶然出现，断线无法自动重连，log如下：
[10:04:50.310]收←◆[1;33m[inf] iotx_mc_cycle(1925): receive ping response!
[0m
[10:05:20.555]收←◆[1;33m[inf] iotx_mc_keepalive_sub(3056): send MQTT ping...
[0m
[10:05:50.812]收←◆[1;33m[inf] iotx_mc_keepalive_sub(3056): send MQTT ping...
[0m
[10:06:12.740]收←◆[0;32mI (56198205) wifi: bcn_timout,ap_probe_send_start[0m

[10:06:15.245]收←◆[0;32mI (56200709) wifi: ap_probe_send_over, rest wifi status to disassoc[0m
[0;32mI (56200711) wifi: state: 5 -> 0 (100)[0m
[0;32mI (56200713) wifi: pm stop[0m
[0;31mE (56200718) conn_mgr: Disconnect reason : 200[0m
[0;31mE (56200727) esp-tls-wolfssl: read error :-397:[0m
[0;31mE (56200733) iot_import_tls: esp_tls_conn_read error, errno:Software caused connection abort[0m
[1;31m[err] IOT_MQTT_Yield(3281): error occur rc=-14
[0m[1;31m[err] iotx_mc_keepalive(2655): network is disconnected!
[0m[1;33m[inf] iotx_cloud_conn_mqtt_event_handle(87): disconnected,fd = 0
[0m[1;33m[inf] _dm_client_event_cloud_disconnect_handle(139): IOTX_CM_EVENT_CLOUD_DISCONNECT
[0m[1;31m[err] IOT_MQTT_Yield(3281): error occur rc=-27
[0m
[10:06:15.382]收←◆[0;31mE (56200846) conn_mgr: Disconnect reason : 201[0m

[10:06:15.442]收←◆[1;31m[err] IOT_MQTT_Yield(3281): error occur rc=-27
[0m
[10:06:15.552]收←◆[1;31m[err] IOT_MQTT_Yield(3281): error occur rc=-27
[0m
[10:06:15.652]收←◆[1;33m[inf] _iotx_linkkit_event_callback(237): Receive Message Type: 1
[0m[1;32;40muser_disconnected_event_handler.89: Cloud Disconnected[0m

[10:06:15.782]收←◆[1;31m[err] IOT_MQTT_Yield(3281): error occur rc=-27
[0m
[10:06:15.892]收←◆[1;31m[err] IOT_MQTT_Yield(3281): error occur rc=-27
[0m
[10:06:16.002]收←◆[1;31m[err] IOT_MQTT_Yield(3281): error occur rc=-27

麻烦提供一下解决方法
谢谢！

yuanjianmin, 2020-08-31:
看起来是WIFI Beacon超时了,可能原因是内存不足或者WIF挂了,请问你使用的SDK是哪个?

bestkeen, 2020-09-01:
ESP8266_RTOS_SDK v3.3-rc1，目前已经解决了，是wifi被挤掉线/AP信号不稳定，导致无法重连，我这边直接重启设备