espressif/esp-ali-smartliving #11: ESP8266开发板采用天猫精灵零配功能失败，使用样例smartlight

tracediary, 2020-10-30:
## 1. 开发环境

- 模组或开发板：ESP8266
- esp-ali-smartliving 版本 1.6：
- ESP8266_RTOS_SDK 版本 3a7964ba (release/v3.3)：
- ESP-IDF 版本 15cc594：

## 2. 问题描述

编译完成smartlight后，下载到ESP8266开发板上，烧录完成三元组，第一次上电，设备自动进入配网状态，在扫描WIFI通道，语音控制天猫精灵找队友功能，可以发现开发板的存在，并且看日志，开发板已经获取到了SSID和password，但是最后连入阿里云后台失败

### 2.1 复现步骤

1. 编译完sdk后，进入smartlight代码目录，编译并下载  make flash monitor
2. 通过天猫精灵语音控制找队友功能，自动寻找该设备，然后在提示找到智能灯设备之后让天猫精灵连接该设备

### 2.2 复现问题的代码

// 如果需要提交复现问题的代码, 可通过 [GIST](https://gist.github.com) 或通过附件的方式提交。

## 3. 调试 Logs
load 0x40100000, len 7168, room 16 
tail 0
chksum 0xe3
load 0x3ffe8408, len 24, room 8 
tail 0
chksum 0x6a
load 0x3ffe8420, len 3548, room 8 
tail 4
chksum 0x17
I (46) boot: ESP-IDF v3.3-31-g3a7964ba-dirty 2nd stage bootloader
I (46) boot: compile time 02:02:10
I (48) qio_mode: Enabling default flash chip QIO
I (55) boot: SPI Speed      : 40MHz
I (61) boot: SPI Mode       : QIO
I (67) boot: SPI Flash Size : 2MB
I (73) boot: Partition Table:
I (79) boot: ## Label            Usage          Type ST Offset   Length
I (90) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (102) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (113) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (125) boot:  3 ota_0            OTA app          00 10 00010000 000f0000
I (136) boot:  4 fctry            WiFi data        01 02 00100000 00004000
I (148) boot:  5 ota_1            OTA app          00 11 00110000 000f0000
I (160) boot: End of partition table
I (166) boot: No factory image, trying OTA 0
I (174) esp_image: segment 0: paddr=0x00010010 vaddr=0x40210010 size=0x9844c (623692) map
0x40210010: _stext at ??:?

I (397) esp_image: segment 1: paddr=0x000a8464 vaddr=0x402a845c size=0x1d500 (120064) map
I (438) esp_image: segment 2: paddr=0x000c596c vaddr=0x3ffe8000 size=0x008c8 (  2248) load
I (439) esp_image: segment 3: paddr=0x000c623c vaddr=0x40100000 size=0x00a40 (  2624) load
I (449) esp_image: segment 4: paddr=0x000c6c84 vaddr=0x40100a40 size=0x059cc ( 22988) load
I (470) boot: Loaded app from partition at offset 0x10000
I (491) system_api: Base MAC address is not set, read default base MAC address from EFUSE
I (497) system_api: Base MAC address is not set, read default base MAC address from EFUSE
phy_version: 1163.0, 665d56c, Jun 24 2020, 10:00:08, RTOS new
I (563) phy_init: phy ver: 1163_0
I (566) reset_reason: RTC reset 1 wakeup 0 store 0, reason is 1
W (571) wrapper_kv: nvs get blob q_rt failed with 1102
E (579) wrapper_kv: Set q_rt blob value
I (588) factory_rst: quick reboot times 1, don't need to restore
I (597) gpio: GPIO[0]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (611) gpio: GPIO[2]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (625) gpio: GPIO[4]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (639) pwm: --- PWM v3.2

[prt] log level set as: [ 4 ]
W (663) wrapper_kv: nvs get blob stassid failed with 1102
[crt] enable awss

I (668) app main: IOTX_AWSS_ENABLE
I (671) app main: IOTX_AWSS_START
[crt] __awss_start

[inf](198) zconfig_init(261): zconfig_init
I (701) awss: Open monitor mode

[crt] chan 7

[crt] chan 8

[crt] chan 9

[crt] chan 10

[inf](21226) zconfig_callback_channel_locked(139): channel lock @ 10
I (21708) app main: IOTX_AWSS_LOCK_CHAN
I (21711) app main: IOTX_AWSS_GOT_SSID_PASSWD
[inf](21236) iotx_guider_set_dynamic_region(298): no region from app so clear region in kv
E (21730) wrapper_kv: nvs erase key mqtt_url failed with 1102
E (21740) wrapper_kv: nvs erase key regionid failed with 1102
***ERROR*** A stack overflow in task ppT has been detected.
abort() was called at PC 0x40223c07 on core 0
0x40223c07: vApplicationStackOverflowHook at /home/jason/esp/ESP8266_RTOS_SDK/components/freertos/port/esp8266/port.c:393

Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x4023c39a  PS      : 0x00000033  A0      : 0x4023c398  A1      : 0x3ffe8490  
0x4023c39a: abort at /home/jason/esp/ESP8266_RTOS_SDK/components/newlib/newlib/port/syscall.c:168 (discriminator 1)

0x4023c398: abort at /home/jason/esp/ESP8266_RTOS_SDK/components/newlib/newlib/port/syscall.c:168 (discriminator 1)

A2      : 0x00000000  A3      : 0x00000001  A4      : 0xffffffdb  A5      : 0x00000001  
A6      : 0x00000000  A7      : 0x3ffec2c0  A8      : 0x00000003  A9      : 0x00000090  
A10     : 0x3fff33b4  A11     : 0x00000027  A12     : 0x00000080  A13     : 0x00000007  
A14     : 0x00000080  A15     : 0x3ffea920  SAR     : 0x0000001e  EXCCAUSE: 0x0000001d  

Backtrace: 0x4023c39a:0x3ffe8490 0x40223c0a:0x3ffe84a0 0x40101150:0x3ffe84b0 0x40100cc3:0x3ffe84c0 0x40100f16:0x3ffe84d0 0x401003bc:0x3ffe84f0 
0x4023c39a: abort at /home/jason/esp/ESP8266_RTOS_SDK/components/newlib/newlib/port/syscall.c:168 (discriminator 1)

0x40223c0a: vApplicationStackOverflowHook at ??:?

0x40101150: vTaskSwitchContext at /home/jason/esp/ESP8266_RTOS_SDK/components/freertos/freertos/tasks.c:4554

0x40100cc3: SoftIsrHdl at /home/jason/esp/ESP8266_RTOS_SDK/components/freertos/port/esp8266/port.c:393

0x40100f16: _xt_isr_handler at /home/jason/esp/ESP8266_RTOS_SDK/components/freertos/port/esp8266/port.c:393

0x401003bc: _xt_user_entry1 at /home/jason/esp/ESP8266_RTOS_SDK/components/freertos/port/esp8266/xtensa_vectors.S:700。

## 4. 其他项

// 如果涉及到配网或其他兼容性问题，请提供具体手机型号、路由型号或其他信息。

yuanjianmin, 2020-11-02:
看起来是task太小溢出了，你试一下make menuconfig里component config->ESP8266-specific->ppT task stack size调大一点。

tracediary, 2020-11-02:
> 看起来是task太小溢出了，你试一下make menuconfig里component config->ESP8266-specific->ppT task stack size调大一点。

嗯，确实是这个问题。
我是直接下载的官方代码，没有更改，官方代码在这个参数设置上有点问题。原stack size设置为2048，我改后设置为4096，可以自动完成零配功能

closed 2020-11-11 by tracediary