espressif/esp-ali-smartliving #12: 编译报错

hqchen118, 2020-11-02:
## 1. 开发环境

- 模组或开发板：
- esp-ali-smartliving 版本（``git rev-parse --short HEAD`` 来获取 commit id）：15cc594
- ESP8266_RTOS_SDK 版本（``git rev-parse --short HEAD`` 来获取 commit id）：3ad3fd7
- ESP-IDF 版本（``git rev-parse --short HEAD`` 来获取 commit id）：

## 2. 问题描述

编译报错：
/home/hankchen/ESP_8266/esp-ali-smartliving/wrappers/wrapper_ota.c:27:28: fatal error: mbedtls/sha256.h: No such file or directory
compilation terminated.
/home/hankchen/ESP_8266/ESP8266_RTOS_SDK/make/component_wrapper.mk:285: recipe for target 'wrappers/wrapper_ota.o' failed
make[1]: *** [wrappers/wrapper_ota.o] Error 1

yuanjianmin, 2020-11-02:
建议使用release/v3.3或者tag v3.3版本。

hqchen118, 2020-11-02:
模组或开发板：
esp-ali-smartliving 版本（git rev-parse --short HEAD 来获取 commit id）：15cc594
ESP8266_RTOS_SDK 版本（git rev-parse --short HEAD 来获取 commit id）：3a7964ba
ESP-IDF 版本（git rev-parse --short HEAD 来获取 commit id）：

mac系统
mac系统
mac系统

编译sdk-c：报错：
make[1]: *** [/Users/hankchen/esp8266/esp-ali-smartliving/ali-smartliving-device-sdk-c/.O/src/infra/utils/digest/utils_md5.o] Error 126
make[1]: *** Waiting for unfinished jobs....

yuanjianmin, 2020-11-03:
mac系统编译阿里的这个库好像是有问题，你有linux的系统吗？

hqchen118, 2020-11-03:
好的，我用ubuntu试试

closed 2020-11-03 by hqchen118