espressif/esp-ali-smartliving #13: 代码烧写到02dc模组，程序卡死不动

hqchen118, 2020-11-03:
## 1. 开发环境

- 模组或开发板：
- esp-ali-smartliving 版本（``git rev-parse --short HEAD`` 来获取 commit id）：15cc594a
- ESP8266_RTOS_SDK 版本（``git rev-parse --short HEAD`` 来获取 commit id）：3a7964b
- ESP-IDF 版本（``git rev-parse --short HEAD`` 来获取 commit id）：

## 2. 问题描述

程序默认选项编译，烧入02DC模组，感觉卡死了，mnitor 监控停止不输出了

### 2.1 复现步骤

1. ...
2. ...

### 2.2 复现问题的代码

// 如果需要提交复现问题的代码, 可通过 [GIST](https://gist.github.com) 或通过附件的方式提交。

## 3. 调试 Logs

ets Jan  8 2013,rst cause:1, boot mode:(3,0)

load 0x40100000, len 7168, room 16 
tail 0
chksum 0xc9
load 0x3ffe8408, len 24, room 8 
tail 0
chksum 0x40
load 0x3ffe8420, len 3544, room 8 
tail 0
chksum 0x01
I (44) boot: ESP-IDF v3.3-31-g3a7964b 2nd stage bootloader
I (44) boot: compile time 16:49:18
I (45) qio_mode: Enabling default flash chip QIO
I (52) boot: SPI Speed      : 40MHz
I (58) boot: SPI Mode       : QIO
I (64) boot: SPI Flash Size : 2MB
I (71) boot: Partition Table:
I (76) boot: ## Label            Usage          Type ST Offset   Length
I (87) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (99) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (110) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (122) boot:  3 ota_0            OTA app          00 10 00010000 000f0000
I (134) boot:  4 fctry            WiFi data        01 02 00100000 00004000
I (145) boot:  5 ota_1            OTA app          00 11 00110000 000f0000
I (157) boot: End of partition table
I (163) boot: No factory image, trying OTA 0
I (171) esp_image: segment 0: paddr=0x00010010 vaddr=0x40210010 size=0x97e50 (622160) map
0x40210010: _stext at ??:?

I (394) esp_image: segment 1: paddr=0x000a7e68 vaddr=0x402a7e60 size=0x1aa68 (109160) map
I (431) esp_image: segment 2: paddr=0x000c28d8 vaddr=0x3ffe8000 size=0x00928 (  2344) load
I (432) esp_image: segment 3: paddr=0x000c3208 vaddr=0x40100000 size=0x00a70 (  2672) load
I (442) esp_image: segment 4: paddr=0x000c3c80 vaddr=0x40100a70 size=0x059cc ( 22988) load
I (463) boot: Loaded app from partition at offset 0x10000
I (482) system_api: Base MAC address is not set, read default base MAC address from EFUSE

yuanjianmin, 2020-11-09:
请问跑的哪个示例，一直卡死在哪里？

hqchen118, 2020-11-09:
smart_light

yuanjianmin, 2020-11-09:
> 
> 
> smart_light

请问是卡在哪里？最后一句打印是哪个？

hqchen118, 2020-11-09:
是的。我没有烧写三元组，和这个有关吗？

yuanjianmin, 2020-11-09:
会的，要烧录三元组。