espressif/esp-ali-smartliving #14: ESP32-S2编译失败，找不到许多方法和函数；

xuhongv, 2020-11-05:
## 1. 开发环境

- 模组或开发板：
- esp-ali-smartliving 版本：15cc594
- ESP-IDF 版本：cd4fa46f5
- 环境： Linux

## 2. 问题描述

无法正常编译

### 2.1 复现步骤

1. 环境正确搭建；
2. 运行脚本 build.sh

### 2.2 复现问题的Logs

```
/home/xuhongv/.espressif/tools/xtensa-esp32s2-elf/esp-2020r1-8.2.0/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/8.2.0/../../../../xtensa-esp32s2-elf/bin/ld: /mnt/d/ESPRESSIF/Linux_Windows/esp-ali-smartliving/ali-smartliving-device-sdk-c/output/release/lib/libiot_sdk.a(awss_reset.o):(.literal.awss_handle_reset_cloud_reply+0x10): undefined reference to `HAL_Timer_Stop'
/home/xuhongv/.espressif/tools/xtensa-esp32s2-elf/esp-2020r1-8.2.0/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/8.2.0/../../../../xtensa-esp32s2-elf/bin/ld: /mnt/d/ESPRESSIF/Linux_Windows/esp-ali-smartliving/ali-smartliving-device-sdk-c/output/release/lib/libiot_sdk.a(awss_reset.o):(.literal.awss_handle_reset_cloud_reply+0x14): undefined reference to `HAL_Timer_Delete'
/home/xuhongv/.espressif/tools/xtensa-esp32s2-elf/esp-2020r1-8.2.0/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/8.2.0/../../../../xtensa-esp32s2-elf/bin/ld: /mnt/d/ESPRESSIF/Linux_Windows/esp-ali-smartliving/ali-smartliving-device-sdk-c/output/release/lib/libiot_sdk.a(awss_reset.o): in function `awss_handle_reset_cloud_reply':
(.text.awss_handle_reset_cloud_reply+0x20): undefined reference to `HAL_Timer_Stop'
/home/xuhongv/.espressif/tools/xtensa-esp32s2-elf/esp-2020r1-8.2.0/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/8.2.0/../../../../xtensa-esp32s2-elf/bin/ld: (.text.awss_handle_reset_cloud_reply+0x2a): undefined reference to `HAL_Timer_Delete'
/home/xuhongv/.espressif/tools/xtensa-esp32s2-elf/esp-2020r1-8.2.0/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/8.2.0/../../../../xtensa-esp32s2-elf/bin/ld: /mnt/d/ESPRESSIF/Linux_Windows/esp-ali-smartliving/ali-smartliving-device-sdk-c/output/release/lib/libiot_sdk.a(awss_reset.o): in function `awss_stop_report_reset':
(.text.awss_stop_report_reset+0x1c): undefined reference to `HAL_Timer_Stop'
/home/xuhongv/.espressif/tools/xtensa-esp32s2-elf/esp-2020r1-8.2.0/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/8.2.0/../../../../xtensa-esp32s2-elf/bin/ld: (.text.awss_stop_report_reset+0x26): undefined reference to `HAL_Timer_Delete'
/home/xuhongv/.espressif/tools/xtensa-esp32s2-elf/esp-2020r1-8.2.0/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/8.2.0/../../../../xtensa-esp32s2-elf/bin/ld: /mnt/d/ESPRESSIF/Linux_Windows/esp-ali-smartliving/ali-smartliving-device-sdk-c/output/release/lib/libiot_sdk.a(awss.o):(.literal.awss_config_press+0x14): undefined reference to `HAL_Awss_Get_Timeout_Interval_Ms'
/home/xuhongv/.espressif/tools/xtensa-esp32s2-elf/esp-2020r1-8.2.0/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/8.2.0/../../../../xtensa-esp32s2-elf/bin/ld: /mnt/d/ESPRESSIF/Linux_Windows/esp-ali-smartliving/ali-smartliving-device-sdk-c/output/release/lib/libiot_sdk.a(awss.o):(.literal.awss_config_press+0x18): undefined reference to `HAL_Timer_Create'
/home/xuhongv/.espressif/tools/xtensa-esp32s2-elf/esp-2020r1-8.2.0/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/8.2.0/../../../../xtensa-esp32s2-elf/bin/ld: /mnt/d/ESPRESSIF/Linux_Windows/esp-ali-smartliving/ali-smartliving-device-sdk-c/output/release/lib/libiot_sdk.a(awss.o):(.literal.awss_config_press+0x1c): undefined reference to `HAL_Timer_Start'
/home/xuhongv/.espressif/tools/xtensa-esp32s2-elf/esp-2020r1-8.2.0/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/8.2.0/../../../../xtensa-esp32s2-elf/bin/ld: /mnt/d/ESPRESSIF/Linux_Windows/esp-ali-smartliving/ali-smartliving-device-sdk-c/output/release/lib/libiot_sdk.a(awss.o): in function `awss_config_press':
(.text.awss_config_press+0x3): undefined reference to `HAL_Awss_Get_Timeout_Interval_Ms'
(.text.iotx_report_devinfo+0x133): undefined reference to `HAL_GetNetifInfo'
/home/xuhongv/.espressif/tools/xtensa-esp32s2-elf/esp-2020r1-8.2.0/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/8.2.0/../../../../xtensa-esp32s2-elf/bin/ld: /mnt/d/ESPRESSIF/Linux_Windows/esp-ali-smartliving/ali-smartliving-device-sdk-c/output/release/lib/libiot_sdk.a(mqtt_client.o):(.literal.iotx_mc_connect+0x58): undefined reference to `HAL_MutexLock'
/home/xuhongv/.espressif/tools/xtensa-esp32s2-elf/esp-2020r1-8.2.0/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/8.2.0/../../../../xtensa-esp32s2-elf/bin/ld: /mnt/d/ESPRESSIF/Linux_Windows/esp-ali-smartliving/ali-smartliving-device-sdk-c/output/release/lib/libiot_sdk.a(mqtt_client.o):(.literal.iotx_mc_release+0xc): undefined reference to `HAL_MutexDestroy'
/home/xuhongv/.espressif/tools/xtensa-esp32s2-elf/esp-2020r1-8.2.0/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/8.2.0/../../../../xtensa-esp32s2-elf/bin/ld: /mnt/d/ESPRESSIF/Linux_Windows/esp-ali-smartliving/ali-smartliving-device-sdk-c/output/release/lib/libiot_sdk.a(mqtt_client.o):(.literal.IOT_MQTT_Construct+0x5c): undefined reference to `HAL_MutexUnlock'
/home/xuhongv/.espressif/tools/xtensa-esp32s2-elf/esp-2020r1-8.2.0/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/8.2.0/../../../../xtensa-esp32s2-elf/bin/ld: /mnt/d/ESPRESSIF/Linux_Windows/esp-ali-smartliving/ali-smartliving-device-sdk-c/output/release/lib/libiot_sdk.a(mqtt_client.o): in function `iotx_mc_get_next_packetid':
(.text.iotx_mc_get_next_packetid+0xc): undefined reference to `HAL_MutexLock'
/home/xuhongv/.espressif/tools/xtensa-esp32s2-elf/esp-2020r1-8.2.0/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/8.2.0/../../../../xtensa-esp32s2-elf/bin/ld: (.text.iotx_mc_get_next_packetid+0x20): undefined reference to `HAL_MutexUnlock'
/home/xuhongv/.espressif/tools/xtensa-esp32s2-elf/esp-2020r1-8.2.0/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/8.2.0/../../../../xtensa-esp32s2-elf/bin/ld: /mnt/d/ESPRESSIF/Linux_Windows/esp-ali-smartliving/ali-smartliving-device-sdk-c/output/release/lib/libiot_sdk.a(mqtt_client.o): in function `iotx_mc_get_client_state':
(.text.iotx_mc_get_client_state+0x6): undefined reference to `HAL_MutexLock'
/home/xuhongv/.espressif/tools/xtensa-esp32s2-elf/esp-2020r1-8.2.0/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/8.2.0/../../../../xtensa-esp32s2-elf/bin/ld: (.text.iotx_mc_get_client_state+0x10): undefined reference to `HAL_MutexUnlock'
/home/xuhongv/.espressif/tools/xtensa-esp32s2-elf/esp-2020r1-8.2.0/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/8.2.0/../../../../xtensa-esp32s2-elf/bin/ld: /mnt/d/ESPRESSIF/Linux_Windows/esp-ali-smartliving/ali-smartliving-device-sdk-c/output/release/lib/libiot_sdk.a(mqtt_client.o): in function `iotx_mc_set_client_state':
(.text.iotx_mc_set_client_state+0x6): undefined reference to `HAL_MutexLock'
/home/xuhongv/.espressif/tools/xtensa-esp32s2-elf/esp-2020r1-8.2.0/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/8.2.0/../../../../xtensa-esp32s2-elf/bin/ld: (.text.iotx_mc_set_client_state+0x10): undefined reference to `HAL_MutexUnlock'
/home/xuhongv/.espressif/tools/xtensa-esp32s2-elf/esp-2020r1-8.2.0/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/8.2.0/../../../../xtensa-esp32s2-elf/bin/ld: /mnt/d/ESPRESSIF/Linux_Windows/esp-ali-smartliving/ali-smartliving-device-sdk-c/output/release/lib/libiot_sdk.a(mqtt_client.o): in function `iotx_mc_push_subInfo_to':
(.text.iotx_mc_push_subInfo_to+0x22): undefined reference to `HAL_MutexLock'
/home/xuhongv/.espressif/tools/xtensa-esp32s2-elf/esp-2020r1-8.2.0/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/8.2.0/../../../../xtensa-esp32s2-elf/bin/ld: (.text.iotx_mc_push_subInfo_to+0x5a): undefined reference to `HAL_MutexUnlock'
/home/xuhongv/.espressif/tools/xtensa-esp32s2-elf/esp-2020r1-8.2.0/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/8.2.0/../../../../xtensa-esp32s2-elf/bin/ld: (.text.iotx_mc_push_subInfo_to+0x8d): undefined reference to `HAL_MutexUnlock'
/home/xuhongv/.espressif/tools/xtensa-esp32s2-elf/esp-2020r1-8.2.0/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/8.2.0/../../../../xtensa-esp32s2-elf/bin/ld: (.text.iotx_mc_push_subInfo_to+0xde): undefined reference to `HAL_MutexUnlock'
/home/xuhongv/.espressif/tools/xtensa-esp32s2-elf/esp-2020r1-8.2.0/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/8.2.0/../../../../xtensa-esp32s2-elf/bin/ld: /mnt/d/ESPRESSIF/Linux_Windows/esp-ali-smartliving/ali-smartliving-device-sdk-c/output/release/lib/libiot_sdk.a(mqtt_client.o): in function `iotx_mc_read_packet':
(.text.iotx_mc_read_packet+0x13): undefined reference to `HAL_MutexLock'
[... cut ...] ference to `HAL_MutexLock'
/home/xuhongv/.espressif/tools/xtensa-esp32s2-elf/esp-2020r1-8.2.0/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/8.2.0/../../../../xtensa-esp32s2-elf/bin/ld: (.text.IOT_MQTT_Yield+0x286): undefined reference to `HAL_MutexUnlock'
/home/xuhongv/.espressif/tools/xtensa-esp32s2-elf/esp-2020r1-8.2.0/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/8.2.0/../../../../xtensa-esp32s2-elf/bin/ld: (.text.IOT_MQTT_Yield+0x2b3): undefined reference to `HAL_MutexLock'
```

yuanjianmin, 2020-11-09:
请确认是否按照https://github.com/espressif/esp-ali-smartliving#613-%E7%BC%96%E8%AF%91-demo-%E7%A4%BA%E4%BE%8B 这个说明里去修改S2 SDK的代码

xuhongv, 2020-11-09:
- 我是直接跑起  build.sh 脚本的

yuanjianmin, 2020-11-09:
但是对于S2芯片需要按照我上面的链接修改IDF SDK

xuhongv, 2020-11-12:
我已经切换到了 V4.2 分支，还是不行；

```git checkout release/v4.2```

yuanjianmin, 2020-11-12:
> 
> 
> 我已经切换到了 V4.2 分支，还是不行；
> 
> `git checkout release/v4.2`

那你按照https://github.com/espressif/esp-idf/commit/7b9ab1e597f1360f01fa63d56323b1d82205ee75 这个链接修改代码了吗？

closed 2021-01-04 by xuhongv