espressif/esp-ali-smartliving #17: 关于在smartlight工程中OTA升级代码问题

tracediary, 2020-12-16:
## 1. 开发环境

- 模组或开发板：模组或开发板：ESP8266
- esp-ali-smartliving 版本：esp-ali-smartliving 版本 1.6：
- ESP8266_RTOS_SDK 版本：3a7964ba (release/v3.3)
- ESP-IDF 版本：15cc594

## 2. 问题描述

参考完成smartlight后，需要开发OTA功能，发现在smartlight中没有OTA的代码，只有调用接口。

相关函数并没有实现OTA的具体部分，比如：
/** fota event handler **/
static int user_fota_event_handler(int type, const char *version)
{
    char buffer[1025 + 1] = {0};
    int buffer_length = 1025; //must set want read len to len + 1

    /* 0 - new firmware exist, query the new firmware */
    if (type == 0) {
        EXAMPLE_TRACE("New Firmware Version: %s", version);

        if (IOT_Linkkit_Query(EXAMPLE_MASTER_DEVID, ITM_MSG_QUERY_FOTA_DATA, (unsigned char *)buffer, buffer_length) == SUCCESS_RETURN) {
            HAL_Reboot();
        }
    }

    return 0;
}

/* cota event handler */
static int user_cota_event_handler(int type, const char *config_id, int config_size, const char *get_type,
                                   const char *sign, const char *sign_method, const char *url)
{
    char buffer[128] = {0};
    int buffer_length = 128;

    /* type = 0, new config exist, query the new config */
    if (type == 0) {
        EXAMPLE_TRACE("New Config ID: %s", config_id);
        EXAMPLE_TRACE("New Config Size: %d", config_size);
        EXAMPLE_TRACE("New Config Type: %s", get_type);
        EXAMPLE_TRACE("New Config Sign: %s", sign);
        EXAMPLE_TRACE("New Config Sign Method: %s", sign_method);
        EXAMPLE_TRACE("New Config URL: %s", url);

        IOT_Linkkit_Query(EXAMPLE_MASTER_DEVID, ITM_MSG_QUERY_COTA_DATA, (unsigned char *)buffer, buffer_length);
    }

    return 0;
}

只有查询，请问如果要集成OTA功能是否要把样例程序中ota_example_mqtt的代码整合到一起？

如果在smartlight工程中已经实现了OTA的功能，请问在哪儿可以查看？

或者说这两个函数已经在SDK中实现了OTA升级？

### 2.1 复现步骤

NA

### 2.2 复现问题的代码

NA

## 3. 调试 Logs

NA

## 4. 其他项

NA

xuhongv, 2020-12-18:
- 参考这个笔记：https://blog.csdn.net/xh870189248/article/details/103779637

yuanjianmin, 2020-12-18:
不需要，底层已经给你封装好了，具体的OTA流程是：平台下发升级任务后会触发调用到user_fota_event_handler函数->然后将升级任务通过IOT_Linkkit_Query(EXAMPLE_MASTER_DEVID, ITM_MSG_QUERY_FOTA_DATA, (unsigned char *)buffer, buffer_length)传递给底层->iotx_dm_fota_perform_sync((char *)payload, payload_len);->dm_fota_perform_sync->在上面这个函数里，会通过https/http下载固件，然后通过HAL_Firmware_Persistence_Write将固件写入设备。所以不需要ota_example_mqtt整合一起，示例smartlight默认支持OTA，你可以试一下。

closed 2021-04-23 by tracediary