espressif/esp-ali-smartliving #19: esp-ali-smartliving 如何移植 esp-mdf

kust336, 2021-01-11:
你好!   
  请问下esp-ali-smartliving 如何移植到 esp-mdf 实现云智能APP配网及控制?
  bilibili 有一个视频是否已经实现了?  https://www.bilibili.com/video/BV1VZ4y1N7Y4?p=4
  可否提供下demo代码?

yuanjianmin, 2021-01-13:
esp-ali-smartliving 使用的是公版阿里云智能 APP wifi 配网, 支持设备热点, 一键配网等等功能.esp-mdf是使用 mesh 蓝牙配网.
直接使用 esp-ali-smartliving 就可以实现 bilibili 视频中的效果. 可以直接参考 该工程下 smart_light 示例.

kust336, 2021-01-13:
程下 smart_light 示例不是mesh的  我们要想用mesh网络的.  单机wifi信号无法传到最远端的设备.

yuanjianmin, 2021-01-14:
你可以参考examples/development_kit/light/main/aliyun_light_example.c里面的逻辑进行移植,conn_mgr_start是配网的入口函数,linkkit_main是连接云平台的入口函数.

kust336, 2021-01-21:
@yuanjianmin 
 你好!
我在尝试移植配网,现在可以通过AP方式配网成功,但会因assertion "0 && "esp_event_loop_create_default()""  重启.
是不是因为conn_mgr_init里面有个  esp_event_loop_init,app_main调用wifi_init里有个esp_event_loop_create_default()冲突了?
代码如下:
   esp_err_t conn_mgr_init(void)
{
    extern esp_err_t HAL_Kv_Init(void);
    HAL_Kv_Init();

    tcpip_adapter_init();

    wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
    ESP_ERROR_CHECK(esp_event_loop_init(conn_mgr_wifi_event_loop_handler, NULL));
    ESP_ERROR_CHECK(esp_wifi_init(&cfg));
    ESP_ERROR_CHECK(esp_wifi_set_storage(WIFI_STORAGE_RAM));
    ESP_ERROR_CHECK(esp_wifi_set_mode(WIFI_MODE_STA));
    ESP_ERROR_CHECK(esp_wifi_start());

    return ESP_OK;
}
//===============================================================
static mdf_err_t wifi_init()
{
    mdf_err_t ret = nvs_flash_init();
    wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();

    if (ret == ESP_ERR_NVS_NO_FREE_PAGES || ret == ESP_ERR_NVS_NEW_VERSION_FOUND) {
        MDF_ERROR_ASSERT(nvs_flash_erase());
        ret = nvs_flash_init();
    }
    MDF_ERROR_ASSERT(ret);

    MDF_ERROR_ASSERT(esp_netif_init());
    MDF_ERROR_ASSERT(esp_event_loop_create_default());
    MDF_ERROR_ASSERT(esp_netif_create_default_wifi_mesh_netifs(&netif_sta, NULL));

    MDF_ERROR_ASSERT(esp_wifi_init(&cfg));
    MDF_ERROR_ASSERT(esp_wifi_set_storage(WIFI_STORAGE_FLASH));
    MDF_ERROR_ASSERT(esp_wifi_set_mode(WIFI_MODE_STA));
    MDF_ERROR_ASSERT(esp_wifi_set_ps(WIFI_PS_NONE));
    MDF_ERROR_ASSERT(esp_mesh_set_6m_rate(false));
    MDF_ERROR_ASSERT(esp_wifi_start());
    return MDF_OK;
}
//=============================================================

串口log如下:

\0x1b[0m\0x1b[1;33m[inf](16491) awss_notify_dev_info(236): coap send notify success

\0x1b[0m\0x1b[0;32mI (17473) conn_mgr: SNTP get time failed (3), retry after 1000 ms
\0x1b[0m
\0x1b[1;33m[inf](17194) awss_notify_dev_info(232): topic:/sys/awss/event/connectap/notify
\0x1b[0m\0x1b[1;33m[inf](17200) awss_notify_dev_info(236): coap send notify success

\0x1b[0m\0x1b[0;32mI (18473) conn_mgr: SNTP get time success
\0x1b[0m
\0x1b[1;33m[inf](18005) awss_notify_dev_info(232): topic:/sys/awss/event/connectap/notify
\0x1b[0m\0x1b[1;33m[inf](18011) awss_notify_dev_info(236): coap send notify success

\0x1b[0m\0x1b[1;33m[inf](18915) awss_notify_dev_info(232): topic:/sys/awss/event/connectap/notify
\0x1b[0m\0x1b[1;33m[inf](18921) awss_notify_dev_info(236): coap send notify success

\0x1b[0map_config.router_ssid : yf2; password : wzc123456789; mesh_id esp_me;channel 0 +++++++++++++++++++++++++++++++
kv_init_dome=======================
\0x1b[0;33mW (20174) wrapper_kv: nvs get blob stassid failed with 1102\0x1b[0m
kv_init_dome=======================
\0x1b[0;33mW (20183) wrapper_kv: nvs get blob scmode failed with 1102\0x1b[0m
\0x1b[0;32mI (20190) gpio: GPIO[0]| InputEn: 1| OutputEn: 1| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 \0x1b[0m
\0x1b[0;32mI (20199) gpio: GPIO[2]| InputEn: 1| OutputEn: 1| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 \0x1b[0m
\0x1b[0;32mI (20208) gpio: GPIO[4]| InputEn: 1| OutputEn: 1| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 \0x1b[0m
\0x1b[0;33mW (20218) [aliyu_mesh, 85]: <ESP_ERR_INVALID_STATE> MDF_ERROR_ASSERT failed, at 0x40131a31, expression: wifi_init\0x1b[0m
assertion "0 && "esp_event_loop_create_default()"" failed: file "/home/mdf/Downloads/ali_mesh/main/mesh_with_aliyun.c", line 85, function: wifi_init

因为已经做了router_ssid的保存所以重启后会自动联上阿里云,但看云智能APP无法成功添加设备.
请帮我分析下,谢谢

yuanjianmin, 2021-01-22:
看起来conn_mgr_init里有些功能其实在wifi_init里已经做好了。你可以只去执行 smart_light里start_conn_mgr里的逻辑，起一个配网的task，然后在GOT_IP的event里调用linkkit_main起连云的task。

kust336, 2021-01-25:
屏蔽esp_event_loop_init(conn_mgr_wifi_event_loop_handler, NULL);后可以完成配网.但是配网完成后因为屏蔽了esp_netif_create_default_wifi_mesh_netifs(&netif_sta, NULL);
后面又重启了,LOG如下:
\0x1b[0;31mE (24395) event: system event loop not initialized via esp_event_loop_init\0x1b[0m
I (24397) mesh: [SCAN][ch:1]AP:4, other(ID:0, RD:0), MAP:0, idle:0, candidate:1, root:0, topMAP:0[c:0,i:1][a8:7d:12:40:0b:60]router found<>
I (24405) mesh: 1315[SCAN]init rc[30:ae:a4:6d:0b:05,-38], mine:0, voter:0
I (24411) mesh: [SCAN:13/13+x+]rc[128][30:ae:a4:6d:0b:05,-38], self[30:ae:a4:6d:0b:04,-38,reason:0,votes:1,idle][mine:1,voter:1(1.00)percent:0.90][128,1,30:ae:a4:6d:0b:05]

I (24428) mesh: <nvs>write layer:0
I (24430) mesh: [DONE]connect to router:yf2, channel:1, rssi:-38, a8:7d:12:40:0b:60[layer:0, assoc:0], my_vote_num:1/voter_num:1, rc[30:ae:a4:6d:0b:05/-38/1]
\0x1b[1;33m[inf](23690) awss_notify_dev_info(232): topic:/sys/awss/event/connectap/notify
\0x1b[0m\0x1b[1;33m[inf](23694) awss_notify_dev_info(236): coap send notify success

\0x1b[0mI (24627) wifi:new:<1,1>, old:<1,1>, ap:<1,1>, sta:<1,1>, prof:1
I (25311) wifi:state: init -> auth (b0)
I (25318) wifi:state: auth -> assoc (0)
I (25327) wifi:state: assoc -> run (10)
I (25341) wifi:connected with yf2, aid = 3, channel 1, 40U, bssid = a8:7d:12:40:0b:60
I (25342) wifi:security: WPA2-PSK, phy: bgn, rssi: -38
I (25349) wifi:pm start, type: 0

\0x1b[0;31mE (25350) event: system event loop not initialized via esp_event_loop_init\0x1b[0m
I (25358) mesh: [scan]new scanning time:600ms, beacon interval:300ms
I (25375) wifi:AP's beacon interval = 102400 us, DTIM period = 1
I (25427) mesh: <nvs>write layer:1
\0x1b[0;32mI (25428) [mwifi, 174]: Parent is connected\0x1b[0m
\0x1b[0;32mI (25436) [aliyu_mesh, 191]: event_loop_cb, event: 7\0x1b[0m
\0x1b[0;32mI (25436) [mwifi, 257]: State represents: 0\0x1b[0m
\0x1b[0;32mI (25440) [aliyu_mesh, 199]: Parent is connected on station interface\0x1b[0m
I (25Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

如果不屏蔽esp_netif_create_default_wifi_mesh_netifs(&netif_sta, NULL);  热点配网时,ESP32不会给手机分配IP地址,无法时行配网.
另外请问下配网时的esp_event_loop_init(conn_mgr_wifi_event_loop_handler, NULL);能不能改成用mdf_event_loop_init(event_loop_cb);进行处理呢?