espressif/esp-ali-smartliving #22: 最新的天猫精灵，已经无法进行配网

flfq, 2021-07-11:
操作过程如下：
ubuntu 18.04

```
apt-get install git wget flex bison gperf python python-pip python-setuptools cmake ninja-build ccache libffi-dev libssl-dev dfu-util libusb-1.0-0
apt-get install libncurses5-dev libncursesw5-dev
apt-get python-serial

mkdir -p ~/esp
cd ~/esp
git clone https://github.com/espressif/ESP8266_RTOS_SDK.git
cd ~/esp/ESP8266_RTOS_SDK
git checkout release/v3.3
export IDF_PATH=~/esp/ESP8266_RTOS_SDK

cd ~/esp
wget "https://dl.espressif.com/dl/xtensa-lx106-elf-linux64-1.22.0-100-ge567ec7-5.2.0.tar.gz"
tar zxvf xtensa-lx106-elf-linux64-1.22.0-100-ge567ec7-5.2.0.tar.gz
export PATH=$PATH:/root/esp/xtensa-lx106-elf/bin

python -m pip install --user -r ~/esp/ESP8266_RTOS_SDK/requirements.txt

#cd ~/esp/ESP8266_RTOS_SDK/examples/get-started/hello_world
#make menuconfig

cd ~/esp
git clone https://github.com/espressif/esp-ali-smartliving.git

cd ~/esp/esp-ali-smartliving/ali-smartliving-device-sdk-c
make reconfig (选择SDK平台)
make menuconfig (选择相关功能配置,默认不需要修改,该步骤可以省略)
make (生成相关头文件和库文件)

cd ~/esp/esp-ali-smartliving/examples/solutions/smart_light
make defconfig
make menuconfig

cp  ~/esp/esp-ali-smartliving/config/mass_mfg/single_mfg_config.csv tm.csv

$IDF_PATH/components/nvs_flash/nvs_partition_generator/nvs_partition_gen.py --input tm.csv --output tm.bin --size 0x4000

$IDF_PATH/components/esptool_py/esptool/esptool.py --port /dev/ttyUSB0 write_flash 0x100000 tm.bin
```

错误如下
```
TART
I (1737) udp: success to establish udp, fd=0
I (1742) os: task name is CoAPServer_yield

I (1746) app main: IOTX_AWSS_ENABLE
E (6584) wrapper_kv: nvs erase key q_rt failed with 1102
I (6588) factory_rst: Quick reboot timeout, clear reboot times
```

flfq, 2021-07-11:
重启设备后的日志如下

```
load 0x40100000, len 7168, room 16
tail 0
chksum 0x9b
load 0x3ffe8408, len 24, room 8
tail 0
chksum 0x12
load 0x3ffe8420, len 3544, room 8
tail 0
chksum 0x66
I (44) boot: ESP-IDF v3.3-46-g6c438bce 2nd stage bootloader
I (44) boot: compile time 21:05:57
I (44) qio_mode: Enabling default flash chip QIO
I (52) boot: SPI Speed      : 40MHz
I (58) boot: SPI Mode       : QIO
I (65) boot: SPI Flash Size : 2MB
I (71) boot: Partition Table:
I (76) boot: ## Label            Usage          Type ST Offset   Length
I (88) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (99) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (110) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (122) boot:  3 ota_0            OTA app          00 10 00010000 000f0000
I (134) boot:  4 fctry            WiFi data        01 02 00100000 00004000
I (145) boot:  5 ota_1            OTA app          00 11 00110000 000f0000
I (157) boot: End of partition table
I (163) boot: No factory image, trying OTA 0
I (171) esp_image: segment 0: paddr=0x00010010 vaddr=0x40210010 size=0x98784 (624516) map
0x40210010: _stext at ??:?

I (394) esp_image: segment 1: paddr=0x000a879c vaddr=0x402a8794 size=0x1d538 (120120) map
I (436) esp_image: segment 2: paddr=0x000c5cdc vaddr=0x3ffe8000 size=0x008c8 (  2248) load
I (437) esp_image: segment 3: paddr=0x000c65ac vaddr=0x40100000 size=0x00a40 (  2624) load
I (447) esp_image: segment 4: paddr=0x000c6ff4 vaddr=0x40100a40 size=0x05abc ( 23228) load
I (468) boot: Loaded app from partition at offset 0x10000
I (489) system_api: Base MAC address is not set, read default base MAC address from EFUSE
I (494) system_api: Base MAC address is not set, read default base MAC address from EFUSE
phy_version: 1163.0, 665d56c, Jun 24 2020, 10:00:08, RTOS new
I (566) phy_init: phy ver: 1163_0
I (569) reset_reason: RTC reset 2 wakeup 0 store 3741866321, reason is 2
W (575) wrapper_kv: nvs get blob q_rt failed with 1102
E (584) wrapper_kv: Set q_rt blob value
I (592) factory_rst: quick reboot times 1, don't need to restore
I (601) gpio: GPIO[0]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (615) gpio: GPIO[2]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (629) gpio: GPIO[4]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (643) pwm: --- PWM v3.2

[prt] log level set as: [ 4 ]
W (669) wrapper_kv: nvs get blob stassid failed with 1102
[crt] enable awss

I (675) app main: IOTX_AWSS_ENABLE
[crt] ssid:adh_a1aUrCVgIYJ_0F74EB

I (685) awss: ssid: adh_a1aUrCVgIYJ_0F74EB
I (1687) app main: IOTX_AWSS_START
I (1694) udp: success to establish udp, fd=0
I (1699) os: task name is CoAPServer_yield

I (1703) app main: IOTX_AWSS_ENABLE
I (6571) factory_rst: Quick reboot timeout, clear reboot times
```

flfq, 2021-07-11:
配网时的日志如下

```
 EFUSE
I (498) system_api: Base MAC address is not set, read default base MAC address from EFUSE
phy_version: 1163.0, 665d56c, Jun 24 2020, 10:00:08, RTOS new
I (566) phy_init: phy ver: 1163_0
I (569) reset_reason: RTC reset 4 wakeup 0 store 4, reason is 7
E (575) wrapper_kv: Set q_rt blob value
E (583) wrapper_kv: Set awss.rst blob value
W (592) factory_rst: factory restore
E (596) wrapper_kv: nvs erase key stassid failed with 1102
E (605) wrapper_kv: nvs erase key pswd failed with 1102
E (614) wrapper_kv: Set scmode blob value
I (622) conn_mgr: conn_mgr_set_sc_mode success
I (628) gpio: GPIO[0]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (642) gpio: GPIO[2]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (656) gpio: GPIO[4]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (671) pwm: --- PWM v3.2

[prt] log level set as: [ 4 ]
W (697) wrapper_kv: nvs get blob stassid failed with 1102
[crt] enable awss

I (702) app main: IOTX_AWSS_ENABLE
[crt] ssid:adh_a1aUrCVgIYJ_0F74EB

I (712) awss: ssid: adh_a1aUrCVgIYJ_0F74EB
I (1720) app main: IOTX_AWSS_START
I (1727) udp: success to establish udp, fd=0
I (1732) os: task name is CoAPServer_yield

I (1736) app main: IOTX_AWSS_ENABLE
E (6574) wrapper_kv: nvs erase key q_rt failed with 1102
I (6578) factory_rst: Quick reboot timeout, clear reboot times
```

ESP-YJM, 2021-09-26:
天猫精灵配网是一键配网，你现在配网模式是设备热点配网，所以你需要切换配网模式。